tobiasfaust/SolaxModbusGateway #105: ESWeb Interface mit > V3.3.0

Reinhard-M, 2024-12-24:


Lazgar, 2024-12-24:
Über den webinstall kann man nach dem Flashen das WiFi konfigurieren. Es wird in den neueren Versionen kein AP mehr gestartet.

tobiasfaust, 2024-12-24:
Ich habe auch extra das Wiki dazu angepasst. Steht auch extra in den Release Notes drin ;)

Reinhard-M, 2024-12-24:
Genau das funktioniert bei mir leider nicht. Der ESP32-S2 wird gelöscht und programmiert, dann kommt der "Installation complete!" mit dem "Next" button. Drücke ich "next" kommt statt des "Change WiFi" Popup aber dann das vorherige mit "Logs & Console". Das habe ich mit der 3.3.0-75, 3.3.0-84 und der 3.3.1-85 getestet, immer das gleiche Ergebnis.

Reinhard-M, 2024-12-24:
[image]
Ich habe mal die Chrome Development Konsole mitlaufen lassen, im Anhang das Ergebnis. Vorher wird zwar von "Hard Reset" gesprochen, zumindest mein Device ist dann aber weiterhin im Programmierzustand. Irgendwie habe ich Probleme mit den "Improv" Sachen.

tobiasfaust, 2024-12-24:
Das „muss“ aber funktionieren. Steck mal den esp vom usb ab und wieder an. 
dann den install Button. Gleich im ersten Popup sollte auch das Wifi Setup aufgelistet sein.

falls das auch nicht geht, nimm mal bitte einen Standard esp, ohne s2. Zumindest zum testen….

Reinhard-M, 2024-12-25:
Hallo Tobias,
ich habe es gerade mit einem ESP32-DOWDQ6 getestet und da wird "Improv Serial" problemlos erkannt (und damit auch der Dialog für WiFi angezeigt). Nur beim ESP32-S2 funktioniert es aktuell nicht. Den DOWD ESP kann ich aber nicht verwenden, leider. Eine Idee wie ich das "Improv Serial" auch für den ESP32-S2 zum Laufen bekomme?

tobiasfaust, 2024-12-25:
Oh, das tut mir leid, ist aber auch merkwürdig da laut „esp Web Tools“ ein ESP32-S2 ebenso unterstützt sein soll.
Ich selbst habe kein ESP32-S2, müsste ich mir demnächst mal zulegen zum testen.
Abgesehen davon ist der S2 ein sehr beschnittener ESP32, ich kann dir nur empfehlen, einen echten ESP32  zu besorgen ;)

Reinhard-M, 2024-12-25:
Das ist schon richtig, für die Aufgabe war er bislang aber absolut ausreichend. Ich müsste schon einen ESP32 bekommen der zumindest einen sehr ähnlichen Footprint wie mein ESP32-S2 mini hat, ansonsten bekomme ich das Teil nicht auf mein Board mit dem 485 Treiber und der Stromversorgung. Sind halt so die Handicaps am Rande. Ich suche mal weiter nach einer Lösung. Wenn ich etwas finde poste ich es hier.

tobiasfaust, 2024-12-25:
Probier mal bitte WLED auf dem S2 zu installieren. Dieser benutzt auch die Improv Library.

http://install.wled.me

Reinhard-M, 2024-12-25:
Mit der WLED Seite bekomme ich das Device nicht einmal richtig in den Programmiermode, gibt immer einen Timeout. Es gibt hier https://github.com/esphome/esp-web-tools/issues/515 eine Diskussion zum Thema, anscheinend aber noch nicht offiziell gefixt. Ich muss wohl mit dem Update noch eine Weile warten...

tobiasfaust, 2024-12-26:
Probier mal bitte den web-installer mit dieser Testseite hier:
https://tobiasfaust.github.io/SolaxModbusGateway/index_806.html

Reinhard-M, 2024-12-26:
Leider weiterhin das Problem mit "Improv WiFi Serial not detected". Ich habe schon einige Lösungen aus den verschiedenen User Groups ausprobiert, bei mir funktionierte bislang noch keine. Auch solche die bei anderen bereits explizit die Lösung für den ESP32-S2 brachten. Ich muss mal beim Ali nach Alternativen für dieses Modul suchen. Vielen Dank für den Versuch es zu richten :-)

tobiasfaust, 2024-12-26:
Nur zur Sicherheit, du hast über die Browser Einstellungen den gesamten Cache geleert? 
Laut deinem verlinkten Thread im esp Web Tools sollte es nämlich mit der Version 8.0.6 funktionieren….

Reinhard-M, 2024-12-26:
Alles durchprobiert, andere Browser, andere Rechner, unter anderem Raspi4, Cach löschen, Inkognito Mode. Das Tool mag meine S2 nicht. Warum auch immer. Ich werde mein Glück mit einem anderen ESP32 versuchen.

MagicSven81, 2024-12-26:
habe bei mir gerade auf dem Steckbrett einen Aufbau mit ESP32-S2 nachgebaut und kann das Problem von Reinhard-M bestätigen.
Es ist allerdings kein Problem von Tobias, sondern scheint ein Problem der Improv Library zu sein.

Wenn man etwas googelt stößt man öfters auf dieses Problem beim S2.

Wie von Tobias vorher schon angesprochen würde ich lieber den "original" ESP32 benutzen.

Reinhard-M, 2024-12-26:
Das mit der Improv Lib (da hängt es) ist absolut richtig, kann ich ebenfalls bestätigen. Mit einem ESP32 WROOM funktioniert alles problemlos. Also: Kein Fehler bei Tobias, eventuell das Ticket für Neuigkeiten zur Lib offen lassen. Bis ich eine für mich gute Lösung gefunden habe bleibe ich bei der v3.2.2. Ist aktuell absolut ausreichend.

tobiasfaust, 2024-12-26:
Das ist nicht die ImprovLib, vielmehr die darüberliegende WebTools JavascriptLib von ESPHome.
Macht für euch aber keinen unterschied, es funktioniert leider aktuell für den S2 nicht :(

StefanNouza, 2024-12-28:
Moin!

In der Hauptsache schreibt das improv-System SSID und Passwort in den NVS-(non volatile storage) Speicher des ESPs, um diese Informationen beim Connect dort auszulesen und zuu verwenden.

Um sich ohne improv behelfen zu können, kann die folgende Modifikation in mqtt.cpp eingebracht werden.

`#include "mqtt.h"
#include "modbus.h"

// if IMPROV is not doing the job:
//  - activate and fill the 2 defines, compile, download and run once
//    your wifi-crerdential will be put into NVS-memory like IMPROV does
//  - deactivate the defines, compile and download - now WiFi should work
#define NONIMPROVSSID "myWiFiSSID"
#define NONIMPROVPWD  "myWiFiPassword"

MQTT::MQTT(AsyncWebServer* server, DNSServer *dns, const char* MqttServer, uint16_t MqttPort, String MqttBasepath, String MqttRoot, char* APName, char* APpassword): 
  server(server), 
  dns(dns), 
  improvSerial(&Serial),
  mqtt_root(MqttRoot), 
  mqtt_basepath(MqttBasepath), 
  ConnectStatusWifi(false), 
  ConnectStatusMqtt(false)
{ 
  
  this->subscriptions = new std::vector<String>{};
 
  WiFi.setHostname(this->mqtt_root.c_str());

  // only to put the wifi-credentials into NVS-memory offline without connecting IMPROV by web
  #ifdef NONIMPROVSSID
    #ifdef NONIMPROVPWD
      dbg.println("setting WiFi SSID and password by include");

      Preferences preferences;
      if (preferences.begin("wifi", false)) {             // create or open namespace wifi
        preferences.putString("ssid", NONIMPROVSSID);     
        preferences.putString("password", NONIMPROVPWD);
        preferences.end();
        dbg.println("WiFi credentials saved to NVS");
      } else {
        dbg.println("Failed to write WiFi credentials to NVS");
      }
    #endif
  #endif

#ifdef ESP32  
  WiFi.onEvent(std::bind(&MQTT::WifiOnEvent, this, std::placeholders::_1));
#endif`

Wenn beide #defines mit den eigenen WiFi-Credentials gesetzt sind, das ganze einmal übersetzt, geladen und ausgeführt wurde, sind die nötigen Werte im NVS-Speicher ausfallsicher hinterlegt.

Nun können die beiden #defines auskommentiert werden, danach verhält sich mqtt.cpp wie vorher - aber WiFi-Connect geht.

Gruß,
  Stefan

tobiasfaust, 2024-12-29:
@StefanNouza 
diese Änderung würde ich nicht empfehlen da bei jedem Start immer ins NVS geschrieben wird. Das NVS hat begrenzte Schreibzyklen.
Besser ist der folgende Weg (verfügbar mit dem aktuellen update):
Dem Wiki der gitpod Anleitung folgen
in der platformio.ini die Wifi Daten eintragen unter: WIFISSID / WIFIPASSWORD
Dann kompilieren und manuell den ESP flashen

StefanNouza, 2024-12-29:
@tobiasfaust 
da hatte ich wohl nicht deutlich genug geschrieben:
- einmal aktivieren, laden, laufen lassen
- danach wieder rausnehmen, erneut laden.
Das NVS sollte natürlich nicht überstrapaziert werden.
Die andere Lösung, die Du gerade beschrieben hast, muß ich mir nochmal ansehen und verstehen. Eine Lösung reicht - dann kann mein Code wieder weg.

tobiasfaust, 2024-12-29:
sorry, hatte ich überlesen. Dieser Code gehört in die ImprovWiFi Library hinein.
Ich habe es eben noch einmal analog deinem Vorschlag geändert. Die WiFi Zugangsdaten sollten nun erstmalig nach einem fehlgeschlagenden Leseversuch ins NVS geschrieben werden wenn diese angegeben wurden. Danach ist auch OTA möglich.
In meinem ersten Commit war OTA nicht möglich da ich die Zugangsdaten eigentlich nicht speichern wollte.

Bitte testet es einmal.

Reinhard-M, 2024-12-29:
Ich habe das Vorgehen getestet und es funktioniert. Allerdings ist mir beim Compile aufgefallen, dass das "littlefs.bin" nicht mehr erzeugt wird. Ich habe das entsprechende File aus einem vorherigen Lauf verwendet, zumindest gab es dann beim Upload keine Probleme. Unter "Files" werden jetzt aber nicht mehr aktuelle Änderungen angezeigt. Obwohl ich die 3 "Wh" Werte deaktiviert und gespeichert habe und im Status nicht mehr sehe werden sie im modbusitemconfig.json weiterhin mit "1" angezeigt. Gleiches gilt für die baseconfig.json Daten. Irgendwie werden nicht die Daten angezeigt die tatsächlich verwendet werden. Ein OTA Update hat leider keine Besserung gebracht.

Reinhard-M, 2024-12-29:
Update: Jetzt geht leider kein OTA Update mehr, keine Ahnung warum...

tobiasfaust, 2024-12-29:
ist leicht zu erklären:
wenn du im gitpod selbst kompilierst muss die littlefs.bin immer separat erzeugt werden. Siehe wiki:

> If you need a recompilation or you have been changed your WiFi Credentials, please type in the command platformio run and platformio run --target buildfs

Wenn du Werte anpasst werden diese zwar übernommen, aber unter Files nicht angezeigt, schuld ist dein Browser-cache. Einfach den Browser-cache löschen und schon siehst du die Änderungen. Normalerweise muss man aber die dateien nicht händisch anpassen. Das ist eher eine Advanced option ;)

StefanNouza, 2024-12-30:
Moin @tobiasfaust,

Deine neue Methode, über build_flags die improv-lib zu "überzeugen", habe ich eben an einem jungfräulichen (noch eingeschweißten) ESP32 DevKitC geprüft - klappt super!
Wenn ich gesehen hätte, daß die Improv-Lib auch von Dir ist, hätte ich den Code auch dort untergebracht. Ich dachte nur: fremder Code -> Lösung muß im eigenen Projekt bewirkt werden -> MQTT macht WiFi-Connect, also dort einbringen... aber so ist es eine runde Sache.

Reinhard-M, 2024-12-30:
Update zum OTA:
Ich hatte noch "Build: 85" auf meinem ESP, damit war ein OTA Upload nicht möglich. Spätestens mit Build 92 geht es problemlos, vielen Dank für die Unterstützung :-)

tobiasfaust, 2024-12-30:
vielen dank :)
ich freue mich immer über eine kleine spende um die Vollversionen von ElegantOTA und WebSerial einkaufen und integrieren zu können ;)

@StefanNouza, die ImprovLib ist nicht von mir, ich habe mir nur einen Fork gezogen und stark erweitert bzgl WiFi Handling und WiFi Credentials

closed 2025-01-02 by tobiasfaust