tobiasfaust/SolaxModbusGateway #106: Fehlende Erläuterung zu Register position

benhubert, 2024-12-25:
Ich möchte weitere Register zur Solax X3 Konfiguration hinzufügen und versuche momentan den Inhalt der Datei zu verstehen. Leider gibt mir die Wiki auch nicht mehr Aufschluss.

Konkret bin ich verwundert über die Angaben in `position`, z.B. folgenden Block:
```
                {
                    "position": [
                        3,
                        4,
                        5,
                        6,
                        7,
                        8,
                        9,
                        10,
                        11,
                        12,
                        13,
                        14,
                        15,
                        16
                    ],
                    "name": "InverterSN",
                    "realname": "Inverter SerialNumber",
                    "datatype": "string"
                },
```

In der Modbus Doku, die ich von Solax erhalten habe, steht, dass die `InverterSN` in den Feldern `0x0000-0x0006` zu finden sind, 14 Chars bzw. 7 Felder vom Typ `uint16`.

Von der Anzahl her passt es, aber wieso startet der Array bei `3` und nicht bei `0`? Ähnlich ist es bei den darauffolgenden Feldern.

Wäre super, wenn mir das jemand erläutern könnte.

tobiasfaust, 2024-12-25:
Hi
Doch, steht im Wiki: ist hier die Nummer 5: https://github.com/tobiasfaust/SolaxModbusGateway/wiki/adding-new-inverters-or-items#find-out-live-data-get-new-items

Lazgar, 2024-12-25:
Hallo benhubert, ich betreue das json für den solax x3, wenn du weitere Werte brauchst kann ich die gerne ergänzen. Was brauchst du?

benhubert, 2024-12-25:
Ok, d.h. die ersten drei Werte sind Header. Diesen Teil habe ich nicht gesehen, Danke.

Mein Ziel ist, das Lade-Verhalten des Akkus auszulesen und in weiterer Folge auch zu steuern. Konkret geht es um die `0x03` Holding-Register
 * `0x008B` für den Lade-Modus
 * und idealerweise auch alle weiteren Register bis `0x00A8`

Für die Steuerung benötige ich aus den `0x06` Write Register vorerst zumindest
 * `0x001f` um den Charge-Mode zu wechseln

Wichtig ist, dass das Register wirklich nur dann geschrieben wird, wenn sich der Wert über MQTT geändert hat, da es ins EEPROM schreibt.

Soweit ich verstehe, müsste ich dafür auch auf irgendeine Weise das Passwort über Modbus übermitteln. Passiert das über das Write Register `0x0000`? Für mich wäre es ok, wenn man einfach dieses Register wie alle anderen Register über MQTT verfügbar macht, sodass sich das Passwort von einer externen Software über MQTT übermitteln lässt.

Lazgar, 2024-12-25:
Ich arbeite gerade daran das json für den x3 zu erweitern inkl. Einiger sinnvoller Set Befehle. Bisschen Zeit brauch ich noch dafür, aber das wird kommen 👍

benhubert, 2024-12-25:
Oh super! Du kannst mir gerne einen Pull Request schicken sobald du etwas hast, dann kann ich gerne testen.

JensHoeppner, 2024-12-30:
@Lazgar Ich fände die Erweiterung um Set-Befehle auch toll

Lazgar, 2024-12-31:
Hier eine Testversion. Alle Einstellungen können bei mir gesetzt werden **ohne** vorher ein Unlock zu senden.

[Solax-X3-test.txt](https://github.com/user-attachments/files/18280690/Solax-X3-test.txt)

In Fhem habe ich folgendes im Attribut setList:

**OperationMode:SelfUse,Feedin,Backup,Manuel,PeakShaving,TOU {"KG/Wechselrichter/SolaX-X3/set/OperationMode ".SolaxValueMapping($EVTPART1)}
ManuelMode:StopChrge&Dischrge,ForceCharge,ForceDischarge {"KG/Wechselrichter/SolaX-X3/set/ManuelMode ".SolaxValueMapping($EVTPART1)}
ExportLimit:slider,0,100,30000 {"KG/Wechselrichter/SolaX-X3/set/ExportLimit ".$EVTPART1 * 0.1}
OffGridMinimumSoC:slider,10,1,25 {"KG/Wechselrichter/SolaX-X3/set/OffGridMinimumSoC ".$EVTPART1}
PhasePowerBalance:on,off {"KG/Wechselrichter/SolaX-X3/set/PhasePowerBalance ".SolaxValueMapping($EVTPART1)}
UnlockSettings:noArg {"KG/Wechselrichter/SolaX-X3/set/UnlockSettings ". ReadingsNum("MQTT2_SolaX_X3_E438BCD8", "InverterAdminPassword", 0000)}**

Fürs übersetzen in Werte die der Wechselrichter versteht, habe ich mir in 99_myUtils.pm das folgende definiert:
**sub SolaxValueMapping($){
 my ($value)=@_;
 my $ret;
 if ($value eq "SelfUse" or $value eq "StopChrge&Dischrge" or $value eq "off")
  {$ret = 0;}
 elsif ($value eq "Feedin" or $value eq "ForceCharge" or $value eq "on")
  {$ret = 1;}
 elsif ($value eq "Backup" or $value eq "ForceDischarge")
  {$ret = 2;}
 elsif ($value eq "Manuel")
  {$ret = 3;}
 elsif ($value eq "PeakShaving")
  {$ret = 4;}
 elsif ($value eq "TOU")
  {$ret = 5;}  
 return $ret;
}**

Nach einem Neustart dauert es ein bisschen bis er Befehle ausführt, und einige können nur verarbeitet wenn der Wechselrichter läuft.
Falls ihr zusätzliche Einstellungen benötigt oder selber welche einbaut, lasst es mich wissen.
Ist auf jeden Fall noch ausbaufähig :)

P.s.: minimum Interval für die Live Daten sind 5 Sekunden.

Lazgar, 2025-01-01:
Ich habe noch den LockState von den Settings mit rein genommen. Bei mir steht der eigentlich immer auf Unlocked... Aber laut Doku muss man die Settings entsperren bevor man Befehle schickt.

[Solax-X3-test.json](https://github.com/user-attachments/files/18284753/Solax-X3-test.json)

@tobiasfaust Wenn im json der letzte Value bei den Live-Daten ein Mapping hat, schmiert mir der ESP permanent ab. Kannst du das nachstellen?

tobiasfaust, 2025-01-04:
Man imho den Inverter nur einmal entsperren, sprich: das passwort in register 0 schreiben. Danach nie wieder.
@Lazgar Mit dem Absturz bei einem Mapping muss ich noch testen. Wenn du es reproduzieren kannst, mach bitte ein neues Issue dazu auf damit ich den überblick behalte ;)

Lazgar, 2025-01-09:
> @Lazgar Mit dem Absturz bei einem Mapping muss ich noch testen. Wenn du es reproduzieren kannst, mach bitte ein neues Issue dazu auf damit ich den überblick behalte ;)

@tobiasfaust Ich konnte den Fehler nicht mehr reproduzieren, lag wohl an mir :D

JensHoeppner, 2025-01-10:
@Lazgar 
Ich habe getestet und komme nicht weiter. Liegt aber mehr an mir als an deiner Programmierung.
Was ich gemacht habe:
Neuen ESP mit aktueller Version 3.3.1@stable geflasht 
Über den Menüpunkt Files die Solax-X3-test.json Datei angelegt und gefüllt, 
Im Menü Modbus Config Solax-X3-test als Inverter-Typ ausgewählt, Set Commands auf enable
in der BasisConfig lautet mein Device Name "SolaX-X3" der MQTT Topic Base "KG/Wechselrichter"
Im FHEM wurde ein Gerät SolaX_X3_5CF6B870 angelegt, Readings erscheinen und werden aktualisiert.
bei diesem Gerät das Attribut setList ausgefüllt wie von dir oben angegeben
Muss dabei außer dem Gerätenamen in der letzten Zeile noch etwas angepasst werden? 
`UnlockSettings:noArg {"KG/Wechselrichter/SolaX-X3/set/UnlockSettings ". ReadingsNum("SolaX_X3_5CF6B870", "InverterAdminPassword", 0000)}`
Muss das Passwort geändert werden? In der App ist bei mir ein anderes Hinterlegt. Oder haben die nichts miteinander zu tun?
die Routine SolaxValueMapping in der 99_myUtils.pm ist auch erstellt.
im Reading InverterLockState wird 2 angezeigt.  
Habe versucht den OperationMode zu ändern, hat leider nicht funktioniert.
Über die App habe ich den OperationMode auf Manual Mode geändert und dann über FHEM versucht ForceCharge oder StopChrge zu aktivieren. Leider auch ohne Erfolg.
Habe ich etwas übersehen? Muss ich noch etwas anpassen?

Lazgar, 2025-01-11:
Das sieht schon sehr gut aus.
ReadingsNum zielt darauf ab das Admin Passwort aus einem Reading zu holen das "InverterAdminPassword" heißt.
Wenn er das nicht findet überträgt er 0000 als Passwort.

Du musst nur im Gateway das Item "Inverter Admin Password" aktivieren, damit wird dir auch das Reading in Fhem angelegt und er überträgt das richtige Passwort.
[image]

Warum beim Settings Lock eine 2 kommt, kann ich nicht nachvollziehen, laut Doku kann da nur 0 oder 1 kommen...
Die restlichen Werte passen? Hast du einen X3 G4 oder X3 G3?

JensHoeppner, 2025-01-11:
@Lazgar 
Wir haben einen X3 G4 im Keller und ich kann Teilerfolge vermelden.
Der Haken bei "Inverter Admin Password" war und ist gesetzt, das Reading "Inverter Admin Password" gibt es auch und ist gefüllt. 

Habe nun "set SolaX_X3_5CF6B870 UnlockSettungs" gesendet. Das Reading "InverterLockState" hat sich von "2" auf "Unlocked" geändert. Auch in der Status Ansicht steht unter "Inverter Settings Lock" nun "Unlocked"

Erfolgreich getestet:
Setzen des Betriebsmodus: funktioniert.
Im manuellen Modus kann zwischen Stop / Laden / Entladen gewechselt werden
das Export Limit kann eingestellt werden.
PhasePowerBalance: kann aktiviert /deaktiviert werden

Nicht erfolgreich getestet: 
OffGridMinimumSoC: Ist das in der App EPS-Einstellung MinSoC (%)? Den bekomme ich nicht geändert.

Was nicht funktioniert:
Die Readings "ExportLimit", "OperationMode", "ManuelMode", "PhasePowerBalance" werden nur beim Start des ESP aktualisiert, danach nicht mehr. 
Einige andere Readings werden auch nicht aktualisiert. Bei denen halte ich es jedoch nicht für nötig, da z.B. die Seriennummer sich nicht im Betrieb ändert.

Wünschen würde ich mir noch die Werte Charger - Max. Ladestrom (A), Charger - Max. Entladestrom (A), Pgrid BIAS, Bias Power (W) einstellen zu können.

Was hat es mit "DRMFunction" und "MPPT" auf sich? Die sind neu. Was ist das?
Weißt du was "EChargeTotal" ist?
Was meinst du mit der Funktion "TargetSoC"?

Nachtrag: Nun steht im reading "InverterLockState" wieder "2". Steuern funktioniert trotzdem.

Lazgar, 2025-01-12:
> Nicht erfolgreich getestet:
> OffGridMinimumSoC: Ist das in der App EPS-Einstellung MinSoC (%)? Den bekomme ich nicht geändert.

Ich kann diesen Wert nur setzen wenn der WR Energie produziert, kannst du mal versuchen tagsüber zu setzen?

> Was nicht funktioniert:
> Die Readings "ExportLimit", "OperationMode", "ManuelMode", "PhasePowerBalance" werden nur beim Start des ESP aktualisiert, danach nicht mehr.

Diese Werte werden aus den ID Daten ausgelesen, die werden nur aktualisiert wenn man einen Set Befehl schickt (dauert ein paar sekunden), oder per default jede Stunde, je nachdem was man hier einstellt:
[image]
Wenn du den Intervall kürzer stellst, kommen die auch öfter, kannst du das bestätigen?

> Wünschen würde ich mir noch die Werte Charger - Max. Ladestrom (A), Charger - Max. Entladestrom (A), Pgrid BIAS, Bias Power (W) einstellen zu können.

Ich habe PgridBias und die Charge Max. sowohl als Value und als Set eingebaut, kannst du mal testen:
[Solax-X3-test2.json](https://github.com/user-attachments/files/18388942/Solax-X3-test2.json)

ACHTUNG: Ich habe einige bestehende Values überarbeitet und vereinheitlicht, daher kann es sein, dass du in Fhem deine DOIFs und|oder Notifys anpassen musst damit die wieder funktionieren.
Falls du für den Wechselrichter keine Statistiken erstellt, würde ich empfehlen alle Readings als auch  das Attribut ReadingList in Fhem zu löschen, zu besseren Übersicht.
Ich musste auch alle Set Befehle Umbenennen um für zukünftige Updates kompatibel zu sein.

Für das Attribut SetList verwende ich aktuell das:
[setList.txt](https://github.com/user-attachments/files/18388957/setList.txt)

In die myUtils.pm musste ich noch die Pgrid Bias Values ergänzen:
[myUtils.txt](https://github.com/user-attachments/files/18388964/myUtils.txt)

Leider konnte ich Bias Power werde zum auslesen noch zum schreiben in der Solax Doku finden.
Ich bin mir nicht sicher ob da nicht das Export Limit greift wenn Pgrid Bias aktiviert ist?!?
Wenn du eine Info findest wo das stehen soll, immer her damit :)

> Was hat es mit "DRMFunction" und "MPPT" auf sich? Die sind neu. Was ist das?

Mit der DRM Funktion hat der Energieversorger die möglichkeit deinen WR abzuschalten wenn in der Nachbarschaft zu viel Energie geliefert wird, ich konnte den Befehl über Modbus nicht setzen, nur über die App. Den Set Befehl habe ich entfernt und nur noch den Value drin gelassen. Bin mir auch nicht sicher ob das überhaupt verwendet wird in DE|AT

MPPT steht für `Max Power Point Tracking` soll den Ertrag erhöhen, kannst ja mal Online nachlesen, in der App heißt das ganze GMPPT, leider unterscheidet sich die App zur Modbus Doku, ich muss da selbst noch ein bisschen nachforschen wie man das richtig beschreibt.

> Weißt du was "EChargeTotal" ist?

Ich bin mir da auch nicht sicher, das ist ein Wert der schon im json war als ich es übernommen hab.
In der Doku steht auch nicht viel dazu... nur `EchargeTotal (Inverter AC Port)`
@tobiasfaust weißt du was der Wert genau aussagt?

> Was meinst du mit der Funktion "TargetSoC"?

Für einen Akku ist es eigentlich eher schlecht wenn er andauernd nur ein bisschen entladen wird und dann gleich wieder voll gemacht wird, oder auch nur ein bisschen geladen wird und dann gleich wieder komplett entladen.
Es gibt die Funktion den Akku erst wieder zu laden wenn er z.B.: 10% erreicht hat, oder eben erst wieder zu entnehmen wenn er 100% erreicht hat. sozusagen einen kompletten Entlade und Ladezyklus zu durchlaufen.
Da ich selbst keine Batterie habe, kann ich es nicht testen.

> Nachtrag: Nun steht im reading "InverterLockState" wieder "2". Steuern funktioniert trotzdem.

In dieser Version des json ist der Aufbau der Abfragen ein anderer, wenn du trotzdem noch immer eine 2 bekommst, kannst du mir vielleicht die SW Version deines WR schicken?
Zu finden in der Solax Cloud Webseite bei deinem WR Device:
[image]

JensHoeppner, 2025-01-12:
@Lazgar 

PV-Produktion ist wegen Schnee grade nicht möglich. 
Den Test vom "OffGridMinimumSoC" muss ich dann später nachholen.
Auch "TargetSoC" ist dann wohl zu Zeit etwas schwierig zu testen.

"InverterLockState" habe ich in der neuen JSON-Version nun gar nicht mehr. Auch auf der Statusseite vom ESP nicht.
Ich konnte auch keine Werte mehr ändern.
Bin dann auf das alte JSON zurückgewechselt und habe den Inverter unlocked. 
Dann konnte ich auch mit der V2 die Werte wieder ändern.
Was mir aufgefallen ist:
Setlist V1:
UnlockSettings:noArg {"KG/Wechselrichter/SolaX-X3/set/UnlockSettings ". ReadingsNum("MQTT2_SolaX_X3_E438BCD8", "InverterAdminPassword", 0000)}
Setlist V2:
UnlockSettings:noArg {"KG/Wechselrichter/SolaX-X3/set/**set**UnlockSettings ".ReadingsNum("MQTT2_SolaX_X3_E438BCD8","InverterAdminPassword",0000)}
Absicht oder versehen?

"ExportLimit", "OperationMode", "ManuelMode", "PhasePowerBalance" werden beim Start oder gem. dem Interval for ID Datatransmission abgerufen. Beim absenden von Set Befehlen anscheinend nicht. Habe das Interval for ID Datatransmission testweise auch auf 5 heruntergesetzt. Dann werden quasi durchgehend Daten abgerufen. Schneller als alle 7 Sekunden scheint es nicht zu gehen, wenn man alle Datenpunkte abruft.
Spricht etwas dagegen die ID Daten so oft abzurufen? Kann man die Datenleitung überlasten?

PgridBias habe ich getestet: funktioniert. Ebenso BatChargeMaxCurrent und BatDischargeMaxCurrent. Werte anzeigen und Setting geht.

PgridBias Watt kann nicht gleich dem export limit sein, sondern muss ein separater Wert sein. Ich habe dort unterschiedliche Werte drin stehen. in welchen Registern die stehen habe ich jedoch auch keine Ahnung

Softwareversionen des WR sind mit deinen identisch

Lazgar, 2025-01-12:
@JensHoeppner

Hab eh geschrieben, daß ich einige Values umbenannt hab zwecks Vereinheitlichung, InverterSettingsLock heißt jetzt nur noch InverterSettings. Du musst alle umbenannten Values neu aktivieren unter modbusitemconfig....
auch die set Befehle musste ich umbenennen für zukünftige Updates, haben alle ein set davor bekommen.
setList.txt und myUtils.txt sind passend zur test2.json, funktionieren aber nicht mit der test.json

Zu Pgrid Bias Power habe ich noch eine Idee um den zu finden, komme aber heute nicht mehr dazu. Spätestens morgen Abend geb ich bescheid 🖖

Lazgar, 2025-01-18:
@JensHoeppner 

Ich melde mich wieder zurück, habe direkt mit dem Support von Solax kontakt aufgenommen.
Der Wert Bias Power kann für den Solax X3 G4 derzeit nicht über ModBus gesetzt werden.
Sie waren sich auch nicht sicher ob es überhaupt eine Wirkung hat wenn man es über die App setzt...
Laut diesem Artikel wird die Funktion Bias Power gar nicht vom WR unterstüzt:
https://kb.solaxpower.com/de/solution/detail/ff8080818e7ebc08018e87e97bdd241a

Nichts desto trotz haben sie ein Ticket an die Entwicklung aufgemacht, auf antwort warte ich noch...

JensHoeppner, 2025-01-19:
@Lazgar 
Vielen Dank für deine Rückmeldung.

Hab grade über die App getestet: Hat definitiv Auswirkungen. Funktioniert genau so wie gewünscht.
Die in dem KB-Eintrag angegebenen Parameter habe ich in der App teilweise auch gar nicht. 
Ist aber auch nicht so tragisch. Hauptsache ich kann die Richtung steuern.

benhubert, 2025-01-19:
Hallo! Vielen Dank für euren Einsatz, ich komme leider doch erst jetzt dazu, mir das anzusehen.

Mein Wechselrichter hat erst vor kurzem ein Update bekommen, bei mir sind daher folgende Versionen im Einsatz:
 * ARM-Version: `1.45-1.15`
 * DSP-Version: `1.48`

Bei mir ist zudem ein HS25 Speicher im Einsatz.

Die verwendete Modbus-MQTT-Version ist `3.2.3(@development)`. Die Verwendete Reg-Datei ist `Solax-X3-test2.json` aus [diesem Kommentar](https://github.com/tobiasfaust/SolaxModbusGateway/issues/106#issuecomment-2585684659)

### Battery Items

Die Battery Items lesen bei mir folgendes aus:
```
Battery User SoC: 0%
Battery User SoH: 100%
Battery Target SoC: 0%
```

In den "Benutzereinstellungen" der Solax-App gibt es bei mir folgende Optionen, die aktuell so konfiguriert sind:
```
Eigenverbrauch > Min Soc (%): 50

Priorität Einspeisung > Min Soc(%): 10
Priorität Einspeisung > Charge Battery to (%): 100

Back Up Modus > Min Soc (%): 50
Back Up Modus > Charge Battery to (%): 50
```

Die drei o.g. Items bleiben bei mir immer unverändert, auch wenn ich den Speicher manuell vom Netz laden lasse. Zudem wäre es für meine Automation notwendig, die die genannten Einstellungen über MQTT setzen zu können.

### Setzen der Konfiguration

Die Set-Funktionen scheinen bei mir keine Wirkung zu haben. Ich sende `0` an `solax/solax001/set/setOperationMode`, doch die Solax-Konfiguration bleibt auf `Back Up Modus`.

Der Item `InverterSettings` meldet `Unlocked`. Ich habe auch versucht, das Passwort an `solax/solax001/set/setUnlockSettings` zu schicken, aber auch das hatte keinen Effekt.

(In der Basis Config habe ich `MQTT Topic Base Path: solax` und `Device name: solax001`.)

Irgendeine Idee, woran das liegen könnte?

Lazgar, 2025-01-19:
@benhubert 

Ist jetzt schwierig zu sagen, Set Befehle hast du erlaubt nehme ich an. 
Wenn du ein falsches passwort schickst, sollte er wieder auf Locked springen, vielleicht mal ausprobieren.

Schaust du in Fhem ob sich die Werte ändern oder am Gateway?
Kannst du mir mal dein setList hier posten?

> Die drei o.g. Items bleiben bei mir immer unverändert, auch wenn ich den Speicher manuell vom Netz laden lasse.

Die haben auch nicht wirklich was miteinander zu tun... 
Ich hab dir die 5 Items und Set Befehle mal hier eingebaut:

[Solax-X3-test2.json](https://github.com/user-attachments/files/18470494/Solax-X3-test2.json)

[image]

Hilft aber nicht wenn keine Befehle verarbeitet werden :/

benhubert, 2025-01-19:
`Enable Set Commands over MQTT` ist aktiviert, ja.

FHEM verwende ich nicht. Ich habe meine eigene Software und schicke die Befehle direkt per MQTT. Getestet habe ich das über den MQTT Explorer.

Die Tabelle in deinem Screenshot kann ich in meinem Web-UI aber nicht finden. Ich tippe daher darauf, dass meine Firmware zu alt ist. Einspielen der Firmware über den Update-Button hat bei mir noch nicht geklappt. Ich melde mich nochmal, sobald ich die aktuelle Version in Verwendung habe.

Lazgar, 2025-01-20:
Du kannst dir die aktuelle Firmware auch hier holen und direkt einspielen:

[WebInstaller](https://github.com/tobiasfaust/SolaxModbusGateway/tree/web-installer/firmware/v3.3.2-110-development)

Ich verwende derzeit eine eigene Firmware, die Tabelle kommt in einem zukünftigen Update...aber auf die aktuelle Version updaten kann kein Fehler sein :)

tobiasfaust, 2025-01-24:
Für den X3 gibt es von @lazgar ein aktuelles json Update im repo. 
Einfach am ESP über OTA den aktuellsten Build für Firmware und Filesystem flashen

benhubert, 2025-01-27:
Hi! Letztendlich hat es dann mit der installierten Version geklappt, ich habe nie herausgefunden, was bei meinen ersten Versuchen falsch war.

In der Zwischenzeit bin ich noch auf folgende Dokumentation gestoßen:
https://kb.solaxpower.com/solution/detail/2c9fa4148ecd09eb018edf67a87b01d2
Für die Fernsteuerung sind das meiner Ansicht nach die wesentlichsten Features der Modbus-Schnittstelle. Diese funktionieren auf meinem Wechselrichter, auch wenn sie in der eigentlichen Dokumentation nicht enthalten sind. Diese Remote-Commands haben vor allem den Vorteil, dass sie nicht in das EEPROM schreiben, welches ja nur eine gewisse Anzahl an Schreibzugriffen erlaubt.

Notwendig für diese Remote-Commands ist allerdings das Schreiben in mehrere Register zugleich. Hierfür muss man sich überlegen, wie man das auf MQTT abbildet.

Unabhängig davon habe ich außerdem herausgefunden, dass mein neuer Solax-Dongle bereits Modbus über TCP beherrscht. Daher habe ich spontan ein eigenes kleines Python-Skript für mich entwickelt, dass die Verbindung zwischen Solax und MQTT herstellt. So dankbar ich für all eure Mühe bisher war, mein ESP und dieses Projekt sind in meinem Fall nun nicht mehr notwendig. Vielen Dank dennoch für all eure Mühe!

Die Remote-Command Schnittstelle in meiner eigenen Lösung sieht nun so aus, dass ich die 10 Werte als CSV-Liste in einem MQTT-Topic erwarte. Beispiele:
 * Exit remote command mode: `0,0,0,0,0,0,0,0,0,0`
 * Produce 1000 W to AC: `1,2,-1000,0,120,0,0,0,300,0`
 * Charge 1000 W from AC to battery: `1,2,1000,0,120,0,0,0,300,0`

Diese transformiere ich dann in die entsprechenden 15 Modbus-Felder, wobei manche davon in 32-Bit-Form gebracht werden müssen, andere nicht. Soweit als Inspiration für euch.

JensHoeppner, 2025-02-02:
@Lazgar 
habe nun die aktuellste Version von dir installiert. 
Da müssen doch auch die SetList und das SolaxValueMapping erweitert werden, oder?

Lazgar, 2025-02-02:
Hallo @JensHoeppner,

In der aktuellsten Version von Tobias macht das mapping der esp. Wenn du ein mouseover be topic machst, siehst du welche values man schicken kann... Im klartext 👍

JensHoeppner, 2025-02-14:
@Lazgar "Bescheid" 
Ich krieg es nicht hin die Setlist so umzubauen, dass es wieder funktioniert. Poste doch bitte einmal deine.

Lazgar, 2025-02-15:
@JensHoeppner ich hab die neuen Set Befehle selbst noch nicht getestet / testen können.

Hier die setList von mir...

[setList.txt](https://github.com/user-attachments/files/18809173/setList.txt)

Damit brauchst du die Funktion fürs Mapping nicht mehr und kannst du aus der myUtils wieder löschen.

Bitte um Info wenn was nicht funktioniert.

P.S.: nicht vergessen, jeden Set Befehl den du verwenden möchtest, muss im Gateway aktiviert werden:

[image]

LG Lazgar

JensHoeppner, 2025-02-15:
@Lazgar 
Oh man, in meiner setList war fast alles korrekt. Bis auf UnlockSettings. Und nun rate was der Wechselrichter noch mal habe wollte...
Danke dir.

Einen Fehler habe ich noch gefunden: Es müsste TOUMode heißen, nicht TUOMode
[image]

Lazgar, 2025-02-15:
@JensHoeppner danke für den Hinweis! Hab schon einen PR eröffnet. Sobald der übernommen wurde kommt es mit dem Update des Filesystems

Reinhard-M, 2025-02-22:
Hallo Leute,
ich arbeite ebenfalls gerade an der Solax-X3. @tobiasfaust, in der v3.3.2_stable ist ein Registerfehler für "SolarEnergyTotal". Statt [306,307,304,305] ist bei dieser Version [306,307,304,30**6**] eingetragen. Mit der v3.3.3_development wurde der Fehler gefixt, soll ich dennoch ein Ticket auf machen?
Die neuen ID und Set Werte funktionieren bei mir problemlos. Aktuell erweitere ich die Liste um Werte die mir noch fehlen. @Lazgar, das Ergebnis werde ich dir dann zum Abgleich und zur Integration schicken. Aktuell habe ich aber ein Problem mit der Set Liste: Mehr als 19 Einträge sind nicht möglich. Komme ich über diesen Wert, bleibt die "Item Config" Seite in einer Dauerladeschleife. @tobiasfaust, kannst du mir da weiterhelfen? Es liegt definitiv an der Anzahl der Einträge, nicht an den neuen Einträgen.

tobiasfaust, 2025-02-22:
Hi
Wenn der Fehler in der Development beseitigt ist, brauchst du natürlich keinen Issue mehr aufmachen ;)

Bzgl des Problems mit der ladeschleife ab 19set Items muss ich mir ansehen, dafür könntest du ein eigenes Issue aufmachen damit ich den Überblick behalte.

Lazgar, 2025-02-23:
@Reinhard-M 
Wenn du fertig bist würde ich mich sehr freuen wenn du mir dein json zukommen lassen könntest.

Danke und LG,
Lazgar

Reinhard-M, 2025-02-23:
@Lazgar 
Klar, kein Thema. Ich fülle auch einige Werte mit "info" auf da sie nicht immer selbsterklärend sind. Und wenn ich ein fehlerhaftes Register sehe korrigiere ich das ebenfalls.

Reinhard-M, 2025-02-23:
Eine Frage noch in die Runde: Ich trage an einigen Stellen Integer Wert Mappings ein. Wenn ich z.B. "Disabled" eintrage erscheint in der MQTT Message "disabled". Gilt auch für bereits vorhandene Werte. Werden Binary Werte ausgegeben passiert das nicht.

tobiasfaust, 2025-02-23:
Das lowercase hat den Sinn das User nicht verzweifeln wenn sie ein Set item senden wollen, nicht auf case sensitive achten und dann der Befehl verworfen wird weil dieser (aufgrund case sensitive) verworfen wird. Deshalb alle auf lowercase. 
Allerdings habe ich das Problem und den Test case von Reinhard noch nicht ganz verstanden. :(

Lazgar, 2025-02-23:
@tobiasfaust Kann ich natürlich verstehen und hab ich auch vermutet. Vielleicht wäre es jedoch möglich sich beim umwandeln in Lowercase auf Set Befehle zu beschränken. Derzeit werden auch ID und Live Daten mit Mapping in Lowercase umgewandelt. Beim Mapping für binary wird nicht in Lowercase umgewandelt. Es ist nicht homogen und zwecks der Lesbarkeit wäre ich dafür es nur Firmware-Intern für Set Befehle umzuwandeln...

tobiasfaust, 2025-02-23:
@lazgar, das macht auch Sinn. Ich schaue es mir nochmal an. Damit es nicht untergeht, könntest du ein eigenes issue dafür erstellen? Ich werde die nächsten Tage wahrscheinlich dazu erstmal keine Zeit haben.

Reinhard-M, 2025-02-23:
"Problem" möchte ich es nicht gleich nennen. Ich hatte bei der Weiterverarbeitung tatsächlich auf "Normal" geprüft. Jetzt mache ich grundsätzlich ein "LC" bevor ich prüfe :)
Mir ging es eher wie @Lazgar weswegen ich seinen Vorschlag auch unterstütze. Es verbessert für mich auch die Lesbarkeit. Ein "SelfUseMode" gefällt mir persönlich besser als ein "selfusemode".

Lazgar, 2025-02-23:
@tobiasfaust @Reinhard-M hab einen PR dafür eröffnet, macht nur noch Lowercase bei Set Befehlen.

Reinhard-M, 2025-02-23:
@Lazgar,
ich habe mal meinen aktuellen Status angehängt. Ich habe hauptsächlich "Self Use" und "Chrg&Dischrg Period" Elemente hinzugefügt und getestet. Funktioniert soweit problemlos bei mir, auch die "Set" Kommandos. Ich kann Charge aktivieren und deaktivieren und die Start und Ende Zeiten einstellen. Wenn ich neues hinzufüge melde ich mich.

[Solax-X3.json](https://github.com/user-attachments/files/18931927/Solax-X3.json)

Reinhard-M, 2025-03-02:
Hallo @Lazgar,

nochmals ein Update. Ich habe ein paar Typos entfernt, "PhasePowerBalance" in "PhaseUnbalancedPowerFeed" umbenannt. die "info" upgedatet und die Werte entsprechend der Solax Beschreibung angepasst. 
Das Ganze habe ich in Home Assistant eingebunden und getestet, funktioniert problemlos inklusive Lock/Unlock der Settings. Zur Veranschaulichung habe ich mal einen Screenshot beigefügt.

[Solax-X3.json](https://github.com/user-attachments/files/19043757/Solax-X3.json)

[image]

Lazgar, 2025-03-03:
Vielen Dank @Reinhard-M!

Gib Bescheid wenn du fertig bist, dann stell ich es ein!

LG

Reinhard-M, 2025-03-03:
@Lazgar, das kannst du bereits einstellen, hatte ich vergessen dazu zu schreiben. Mit dem aktuellen JSON funktioniert alles was ich brauche:-)

Reinhard-M, 2025-03-03:
@tobiasfaust, ich gehe davon aus du sprichst mit deinem Post @Lazgar an, richtig?

Lazgar, 2025-03-03:
hab ich eingestellt

Reinhard-M, 2025-03-03:
@tobiasfaust, eine Sache hatte ich vergessen zu fragen. 3 Elemente haben jetzt einen "settopic" Eintrag, welche Wirkung hat dieser? Für die von mir eingetragenen Set Werten habe keinen solchen Eintrag bei den zugehörigen Lesewerten gemacht und keinen Unterschied zu den 3 genannten feststellen können.

Lazgar, 2025-03-03:
das ist noch ein Überbleibsel von meinen eigenen Testversionen. Hab ich mal entfernt...

JensHoeppner, 2025-03-09:
Bei mir wurde letze Woche eine 2. (Fremdhersteller) PV Anlage mit einem weiteren Solax-Zähler eingebunden. Diese wird in der Solax-App und der Webseite ebenfalls angezeigt. Können diese Werte ebenfalls per Modbus ausgelesen werden?  
Habe in der Modbus-Doku dazu nichts gefunden, aber eigentlich müssten die Werte ja da sein. Woher sollte die App/Webseite sonst die Daten haben.

tobiasfaust, 2025-03-10:
Macht mal bitte bei einem neuen Thema auch ein neues Issue auf…. 

Ein neuer WR braucht auch ein eigenes Gateway. Jedes Gateway bearbeitet immer genau ein Modbus Gerät

closed 2025-03-10 by tobiasfaust