tobiasfaust/SolaxModbusGateway #113: Gateway does not run if it is connected to the inverter when booting.

saschoctm, 2025-01-01:
As the title says, when the gateway starts and is connected to the Growatt SPH inverter, the gateway hangs somehow.
- Web server partially unavailable
- When accessible, the pages sometimes do not open or the red loading circles are constantly visible
- no connection to the inverter

If I start the gateway without a connection to the inverter and then connect it, it runs without any problems.

Since the hardware has been running for about a year with an old software version Dec 2023 or Jan 2024 of the gateway without any problems, should I blame the problem on the software? As I said, I haven't updated the gateway for a year, so I can only say that the current version behaves like this and the old version worked without any major problems, sometimes it needs a restart but rarely.
Nevertheless, I tried another ESP32 board and also HW-0519, always the same result.

Does anyone know the problem or can help?

eckhard58, 2025-01-02:
Hallo saschoctm,

ich hatte mit den alten Releases ähnliche Probleme. Mit der neuen Relaises ab 3.3.1 lief bei mir alles stabil. (siehe Bild)
Tobias schrieb mir mal dies:
"Abgesehen davon sind die Dateien in deinem Filesystem veraltet, diese sollten bei einem OTA Update immer zusammen als Firmware/Filesystem Bundle geupdatet werden."
Das könnte ein Grund für deine Probleme sein.

[image]

saschoctm, 2025-01-02:
Hallo eckhard58,

danke für die Nachricht.
Allerdings benutze ich den letzten development Stand und habe den Webinstaller benutzt. 
Da sollte alles korrekt sein.
Wie gesagt wenn der WR erst nach dem booten angeschlossen wird läuft das gateway scheinbar ganz normal.
Irgendwas muss ja beim booten passieren was das Gateway quasi zum Absturz bringt.

eckhard58, 2025-01-02:
Fragst du ev. Speicherbereiche ab die es nicht gibt?
Ich benutze gerade diese Json mit der letzten Relaise.
[Growatt-SPH2.json](https://github.com/user-attachments/files/18289956/Growatt-SPH2.json)
Kannst die gerne mal probieren. Ich habe einen einphasen WR Growatt SPH 3000 Bj. 08/24

saschoctm, 2025-01-04:
Zur Vollständigkeit Problem behoben.
Der ESP ist abgestürzt beim parsen des ID-Daten

Der requestIdData String musste erhöht werden von 0x14--> 0x20 auch dann erst wird die SN korrekt ausgelesen.
Wird zwingend benötigt für einen Growatt  SPH TL3 BH-UP Series

"RequestIdData": [
        [
          "#ClientID",
          "0x03",
          "0x00",
          "0x00",
          "0x00",
          "0x20"
        ]

"id": [
        {
          "position": [
            49,
            50,
            51,
            52,
            53,
            54,
            55,
            56,
            57,
            58
          ],
          "name": "InverterSN",
          "realname": "Inverter SerialNumber",
          "datatype": "string"
        }

closed 2025-01-06 by tobiasfaust