tobiasfaust/SolaxModbusGateway #114: Fehler falsche Topic Openwb Übergabe

saschoctm, 2025-01-04:
Hab noch nen Fehler gefunden, falscher topic
 das muss heissen:
"setbatw": "openWB/set/houseBattery/W"

{
                    "position": [268, 269, 270, 271],
                    "position2": [264, 265, 266, 267],
                    "name": "BatChargingPower",
                    "realname": "Battery Charging Power",
                    **"openwbtopic": "setbatimpwh",**
                    "datatype": "float",
                    "factor": 0.1,
                    "unit": "W"
                },

tobiasfaust, 2025-01-04:
Da hast du mich abgehängt, du schreibst oben setbatw, unten beziehst du dich aber auf setbatimpwh.

was muss jetzt wo genau angepasst werden?

saschoctm, 2025-01-04:
Oh okay sorry,

so muss es sein:
{
"position": [268, 269, 270, 271],
"position2": [264, 265, 266, 267],
"name": "BatChargingPower",
"realname": "Battery Charging Power",
"openwbtopic": "setbatw",
"datatype": "float",
"factor": 0.1,
"unit": "W"
},

closed 2025-01-18 by tobiasfaust