tobiasfaust/SolaxModbusGateway #118: v3.3.1 Build 107 stable lässt sich nicht über "Update" laden

Reinhard-M, 2025-01-10:
Die neue "stable" Version lässt sich nicht über "Update" laden. Weder "Precompiled" noch "Custom" Firmware. Im ersten Fall startet der Upload nicht einmal, im zweiten Fall startet zwar der Upload, endet aber mit einem "Upload failed". Mit der v3.3.1 Build 106 funktioniert es noch einwandfrei. Das habe ich mit einem ESP32 und einem ESP32-S2 getestet. Wenn ich mir über Git eine eigene Version baue und mit dem Flash Toll lade funktioniert es in allen Fällen.

tobiasfaust, 2025-01-10:
Schau ich mir an…. Danke für die Info

tobiasfaust, 2025-01-11:
Fehler gefunden…. Da mault github mit der CORS Police rum. 
Ich muss die stable löschen, einen kleinen fix im JavaScript einspielen und neu hoch stagen als stable.

alternativ muss man die stable manuell aus github laden und als Custom file (Firmware + Littlefs) über ota laden

Reinhard-M, 2025-01-11:
Das Firmware File ließ sich auf diesem Weg problemlos laden, Littlefs aber nicht. Getestet auf dem ESP32-S2.

tobiasfaust, 2025-01-11:
> Das Firmware File ließ sich auf diesem Weg problemlos laden, Littlefs aber nicht. Getestet auf dem ESP32-S2.

Doch, das geht eigentlich. Denk aber dran vorher die SelectList auf Filesystem zu ändern bevor du die littlefs flashen willst.

Der Webinstaller sollte auch funktionieren (inkl. dem Nachteil das alle configs dann auch gelöscht sind)
Nur das OTA Flash hat aktuell das CORS Problem

Reinhard-M, 2025-01-11:
Das hatte ich übersehen, mit der richtigen Selection funktioniert auch der Littlefs Upload :-)
Daraus schließe ich jetzt, dass über "Update" tatsächlich nur das "firmware.bin" und das "littlefs.bin" hochgeladen werden. "bootloader.bin" und "partitions.bin" benötigen kein Update und ich muss an dieser Stelle auch nicht das "merged-firmware..." hochladen. Sorry, das war mir nicht bewusst.

tobiasfaust, 2025-01-11:
Bitte zuerst die Development 108 laden, danach sollte auch OTA der neuen 109 stable funktionieren

Reinhard-M, 2025-01-11:
Build 108 funktioniert perfekt, 109 wird aber noch nicht angeboten. Habe das Gefühl, die Oberfläche reagiert deutlich schneller als früher.

Reinhard-M, 2025-01-11:
Auch die 109 läuft problemlos mit dem S2 :-)

closed 2025-01-18 by tobiasfaust