tobiasfaust/SolaxModbusGateway #120: Sicherheitsabfrage beim Reset

JensHoeppner, 2025-01-12:
Hallo Tobias,

Zuerst einmal ein Lob. Tolles Projekt, hilft mir sehr meine Anlage zu überwachen.

Nun der Verbesserungsvorschlag, da ich grade versehentlich meine Konfiguration zurückgesetzt habe:
Es wäre eventuell von Vorteil, wenn auf der Statusseite beim Klick auf den Button "Reset" nicht sofort der Reset durchgeführt würde, sondern vorher noch eine Abfrage käme. Vom Reboot zu Reset sind nur wenige Millimeter.
Wie das beim Button "WifiReset" ausschaut habe ich nicht ausprobiert.

tobiasfaust, 2025-01-12:
ckeck mal bitte ob es das ist was du meinst. Musst Style.css und status.html manuell aus dem Repo ziehen und über die files WebUI auf deinen ESP ersetzen. Bitte browsercache danach leeren

JensHoeppner, 2025-01-13:
Ja, genauso! Top.