tobiasfaust/SolaxModbusGateway #121: Keine Daten im SOLAX-Gateway

aceron76, 2025-01-17:
Hallo Tobias,  ich habe es mit unterschiedlichen Releases probiert aber es ist immer dasselbe es werden keine Daten im Gateway angezeigt. Über USB mit hterm werden Daten vom WR aber empfangen, deshalb gehe einmal davon aus das die Verkabelung korrekt ist. Aktuell habe ich das Releas 3.3.1. auf dem ESP  , TX/RX habe ich auch schon vertauscht. Ich habe auch alle anderen Issues durchgelesen um irgendeinen Hinweis zu finden was bei meinem Setup vielleicht falsch sein könnte leider ohne Erfolg. Wäre super wenn du mir etwas helfen könntest.
Danke sG Erwin

[image]

eckhard58, 2025-01-17:
Hallo Erwin,

hast du meine Mail im Fhem-Forum gelesen?

_Bei meinem Growatt SPH 3000 einphasig Bj. 08/24 sind es auch Pin 1 und 5.
Bei meinem Sohn sind es die Pin 4 und 5 der Wechselrichter ist ein  Growatt SPH3600 einphasig etwa 3 Jahre alt.
Da kannst du nur beides mal ausprobieren und die Pin´s mal tauschen.

Und am WR in den Settings die RS485 auf VDD stellen._

Gruß
Eckhard

aceron76, 2025-01-19:
Hallo Eckhard, danke für deine Hinweise. Habe alles gelesen und probiert bei meinem WR  growatt sph 10000tl3 bh-up sind es definitiv die Pins 1 und 5 denn über usb und hterm empfange ich Daten.
sG
Erwin

aceron76, 2025-01-19:
Hallo Eckhard, habe das Problem gelöst es war das Board HW-97, mit dem HW-0519 funktionierts auf Anhieb.
Danke für deinen Support
sG
Erwin

[image]

closed 2025-01-19 by aceron76

eckhard58, 2025-01-19:
Was sind das für Boards?Bin grad für eine Woche im Urlaub. Gruß Eckhard

image.png (view on web)

aceron76, 2025-01-19:
Das sind die RS485 boards von der Beschreibung how to wire the hardware

aceron76, 2025-01-19:
Ich habe gelesen das du ein Problem mit mqtt hattest ich habe dasselbe Problem alle ca 5 Minuten taucht im fhem ein neuer solax gateway auf wie konntest du das lösen?
sG Erwin

tobiasfaust, 2025-01-19:
Auf der Basis-Config Seite die dynamische MQTT id abschalten und auf statistische Id umstellen

aceron76, 2025-01-20:
Hallo noch etwas komisches, plötzlich ist das Growatt-SPH.json nicht mehr geladen

[image]