tobiasfaust/SolaxModbusGateway #122: Modbus Befehle an Deye 12k senden

tedesco1968, 2025-01-17:
Hallo,

ist es auch möglich an den Wechselrichter Modbusregister was zu Schreiben? Wenn Ja wie geht das?

Ich würde gerne in die Adresse 143  die Werte ändern.

Leider habe ich trotz Wiki noch nicht verstanden wie man die Werte ausliest ich benutze die Deye_SUN_SG04LP3.json von MagicSven81.

Villeicht kann mir das einer mal anhand der 143 Adresse erklären, dann könnte ich auch die eine oder andere Adresse Aktualisieren bzw.ergänzen.

Gruß Toni

tobiasfaust, 2025-01-18:
Ja das geht, schau mal in einen der anderen issues. Da wird das Thema auch gerade behandelt. 
Schlussendlich musst du das zu beschreibende Register aus deiner Modbus Doku heraussuchen.

tobiasfaust, 2025-01-29:
die neue Solax-X3.json datei ist eine gute vorlage. Hier gibt es jetzt schon sehr viele set commands.
Wenn möglich teile deine fertige deye json datei hier damit ich diese ins repo übernehmen kann

MagicSven81, 2025-02-09:
Hallo zusammen,
seid ihr in diesem Thema schon weitergekommen?
Mir selbst fehlt leider aktuell komplett die Zeit, sonst hätte ich das schon längst mal gemacht.

tobiasfaust, 2025-02-09:
Nicht von mir aus, das kann nur jemand mit einem Deye machen

tedesco1968, 2025-02-09:
Hallo,

ein Deye habe ich, das wäre das Problem nicht. Einer muss mir mal anhand einer Adresse wie das ganze mit dem Auslesen funktioniert da ich das leider trotz Wiki nicht verstanden habe.
Dann könnte ich mich ans das Befehle senden widmen.

Gruß Antonio

tobiasfaust, 2025-02-10:
Du musst dir die Set Register die du beschreiben willst aus der Deye Modbus Doku heraussuchen.
Schau mal in die Set commands vom Solax-X3, ich denke dann wird es klarer….

closed 2025-03-24 by tobiasfaust