tobiasfaust/SolaxModbusGateway #133: Solax X3 Hybrid SolarEnergy Total aktualisiert nicht mehr regelmäßig

JensHoeppner, 2025-02-09:
Hallo,

ich habe ein Problem mit dem Datenpunkt "SolarEnergy Total".
Dieser wird nicht mehr regelmäßig aktualisiert. Das Heißt der Wert steigt bei PV Produktion nicht mehr kontinuierlich, sondern springt irgendwann um einen großen Betrag nach oben.
Sieht für den gestrigen Tag so aus:
[image]

Habe mich heut mal drangesetzt, und direkt auf dem ESP die Werte beobachtet:
"SolarEnergy Total" bleibt immer gleich, "SolarEnergy Today" wächst in 0,1 kWh Schritten.
[image]

Das Problem besteht bei mir seit 18.1.25.
In dem Zeitraum habe ich von einer älteren Version auf neue Version mir Möglichkeit der Steuerung des WR gewechselt.
Heute habe ich auf die aktuelle Stable Version aktualisiert. Problem ist aber noch vorhanden.

Lazgar, 2025-02-09:
Hallo @JensHoeppner 

in welchem Interval fragst du die Live Daten ab?
Mit welcher Version trat das Problem das erste mal auf?

Ich frage die Live Daten alle 10 Sekunden ab und hab solch ein Problem noch in keiner Version gehabt...

LG Lazgar

Lazgar, 2025-02-09:
@JensHoeppner Ich denke ich habe den Fehler gefunden.
@tobiasfaust  Ich mach nen PR mit der kurrektur

JensHoeppner, 2025-02-09:
Wow, das ging schnell.
Danke

closed 2025-02-09 by JensHoeppner

Lazgar, 2025-02-09:
@JensHoeppner tobias hat die Änderung bereits übernommen. Update des Filesystems (Build 121) wird das Problem beheben 👍

JensHoeppner, 2025-02-09:
@Lazgar danke für die Info. Läuft nun wieder korrekt.