tobiasfaust/SolaxModbusGateway #138: Fehler beim Öffnen des Projekts mit platformIo

phant0mias, 2025-02-23:
Hallo,
ich wollte den aktuellen Entwicklungsstand in PlatformIo öffnen und kompilieren.

Es kommt dann folgende Fehlermeldung:
Resolving firmware_ESP32-WebSerial dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing firmware_ESP32-WebSerial (board: esp32dev; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Traceback (most recent call last):
  File "c:\User\Projects\PlatformIO\App\Projekte\SolaxModbusGateway-development\scripts\build_flags.py", line 21, in <module>
    globals()[sys.argv[1]]()
  File "c:\User\Projects\PlatformIO\App\Projekte\SolaxModbusGateway-development\scripts\build_flags.py", line 16, in git_owner
    owner = re.search(r'github.com[:/](.*)/', output.decode()).group(1)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'group'
OSError: 'python scripts/build_flags.py git_owner' exited 1:
  File "C:\Users\werner\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\werner\.platformio\platforms\espressif32\builder\main.py", line 368:
    target_elf = env.BuildProgram()
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\werner\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\werner\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 139:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\werner\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 238:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\werner\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 195:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Environment.py", line 1034:
    do_parse(arg)
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Environment.py", line 882:
    arg = self.backtick(arg[1:])
  File "C:\Users\werner\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Environment.py", line 793:
    raise OSError(f'{command!r} exited {cp.returncode}')
========================== [FAILED] Took 1.86 seconds ==========================

Environment               Status    Duration
------------------------  --------  ------------
firmware_ESP32-WebSerial  FAILED    00:00:01.857
==================== 1 failed, 0 succeeded in 00:00:01.857 ====================

Hat jemand eine Idee was das Problem sein könnte. Muß ich noch etwas installieren.

Das passiert mir erst ab Version 3.3, die vorherigen Versionen kann ich übersetzen

Danke
Werner

tobiasfaust, 2025-02-23:
Du musst das github Projekt auschecken das es auch als git Projekt bei dir auf dem Rechner vorliegt.

Ansonsten benutze gitpod (siehe Wiki)

phant0mias, 2025-02-23:
danke, so funktioniert es

closed 2025-02-23 by tobiasfaust