tobiasfaust/SolaxModbusGateway #139: Installation mit WebInstaller noiht möglich

phant0mias, 2025-02-23:
Hallo,

als Controller habe ich einen ESP32-S3-DevKitC-1 v1.0 verwendet.

Die Installation sämtlicher Entwicklungstools erfolgte über PlatformIo, ich habe keine weiteren Installationen durchgeführt.

Die Installation habe ich mit Microsoft Edge und Google Chrome unter Windows 10 probiert. Auch habe ich mehrere Firmware Images (Stable und Development) ausprobiert. Es ist immer das gleiche Ergebnis.

1.) ESP anschließen
2.) Auswahl von Standard Installation
3) Drücken des Install Buttons
4) Auswahl der Kommunikationsschnittstelle
5) Es wird verbunden
6) Anzeige des Dialogs mit der Möglichkeit von Install und Consolle
7) Install drücken
8) Löshinweisdialog -> install
9) Erasing -> Installing -> Wrapping up -> Installation Complete -> Next
10) UInd dann kommt wieder der Dialog indem nur 2 Optionen angeboten werden (wie bei Punkt 6), nicht wie im Wiki beschrieben ein Dialog indem Solax Einstellungen etc. durchgeführt werden können

Hat jemand eine Idee was ich falsch mache.

phant0mias, 2025-02-23:
Das gleiche Ergebnis habe ich auf einem Windows 11 Rechner

tobiasfaust, 2025-02-23:
Merkwürdig.,  geh mal im webinstaller auf den Button „log“ und dann wieder zurück. Ggf den esp nochmal rebooten.

phant0mias, 2025-02-23:
So hat es auch nicht funktioniert. Evtl. liegt es daran, daß ich auf beiden Rechnern Zadig verwendet habe um für Raspberry Pico COntroller einen anderen Traiber zu installieren. Ich suche einen anderen Rechner der neu installiert ist und probiere es nochmal. Ich melde mich dann

phant0mias, 2025-02-23:
Gleiches verhalten bei einem neu installierten Windows 11 Rechner. Funktioniert auch nicht.

Das ist das Log das in der Console herunterladen kann:

[15:14:19]ESP-ROM:esp32s3-20210327
[15:14:19]Build:Mar 27 2021
[15:14:19]rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
[15:14:19]Saved PC:0x420d344e
[15:14:19]SPIWP:0xee
[15:14:19]mode:DOUT, clock div:1
[15:14:19]load:0x3fce3808,len:0x4bc
[15:14:19]load:0x403c9700,len:0xbd8
[15:14:19]load:0x403cc700,len:0x2a0c
[15:14:19]entry 0x403c98d0
[15:14:19][   256][E][vfs_api.cpp:105] open(): /littlefs/config/baseconfig.json does not exist, no permits for creation
[15:14:19][   267][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
[15:14:21][  2483][E][vfs_api.cpp:105] open(): /littlefs/config/modbusconfig.json does not exist, no permits for creation
[15:14:23][  4104][E][vfs_api.cpp:105] open(): /littlefs/config/modbusitemconfig.json does not exist, no permits for creation

MagicSven81, 2025-02-23:
versuche mal einen "normalen" ESP32

tobiasfaust, 2025-02-23:
Ja, nimm mal bitte einen normalen ESP32
Alternativ gibt es die Hintertür, über gitpod in der platformio.ini einmalig die wifi Zugangsdaten einzutragen und die so gebaute Firmware auf den esp zu flashen. Danach funktioniert auch wieder normales OTA

phant0mias, 2025-02-23:
habe leider ekinen normalen ESP32, nur noch einen Arduino Nano Esp32. Ich schaue mal ob ich einen auftreiben kann.

closed 2025-02-23 by phant0mias