tobiasfaust/SolaxModbusGateway #145: Deye SUN-12K mehrere Akku´s

Sefina-DS, 2025-03-29:
Hi zusammen,

Wir haben mehrere Akkus parallel angeschlossen, die mit dem Deye zuverlässig kommunizieren.
Gibt es eine Möglichkeit sich die Werte der einzelnen Akkus über Modbus auszulesen?

MfG

tobiasfaust, 2025-03-29:
Grundsätzlich: Ja
Aber nur, wenn dein Deye entsprechende Modbus Register besitzt die man auslesen kann

Sefina-DS, 2025-03-29:
[image]

tobiasfaust, 2025-03-29:
Hast du die aktuellste Firmware auf dem deye?
Die Fehler sehen so aus das Zuviel Register abgefragt werden und die Abfrage in eine Range kommt die der deye nicht kennt.

Sefina-DS, 2025-03-29:
Habe Heute frisch installiert (3.3.3).
Ich habe nur wie auf dem Bild die Zahlen eingegeben und abgefragt.

Warum da soviele stehen weiß ich nicht.

tobiasfaust, 2025-03-29:
Dein deye kann wohl die letzten Register nicht.
Da musst du mal selbst anpassen.
Gehe auf die files Seite im WebUi und lade das deye json im Ordner Regs/

Dort steht ganz oben:
` "RequestLiveData": [
                [
                    "#ClientID",
                    "0x03",
                    "0x01",
                    "0xF4",
                    "0x00",
                    "0x77"
                ],
                [
                    "#ClientID",
                    "0x03",
                    "0x02",
                    "0x6B",
                    "0x00",
                    "0x41"
                ]
            ], `

Und ersetze dies durch

` "RequestLiveData": [
                [
                    "#ClientID",
                    "0x03",
                    "0x01",
                    "0xF4",
                    "0x00",
                    "0x77"
                ]
            ], `

Das gesamte json über jsonlint.com nochmal testen und Danach Abspeichern. Dann den esp restarten.

Sefina-DS, 2025-03-29:
Seit dem ich das gemacht habe, obwohl die änderung valide war startet er nicht mehr .....

Ich finde es auch sehr Seltsam das der Gesammtakkustand so deviniert ist :
{
                 "position": [
                        179,
                        180
                    ],
                    "name": "BatteryCapacity",
                    "realname": "Battery Capacity",
                    "datatype": "integer",
                    "unit": "%"
                },

[image]

tobiasfaust, 2025-03-30:
Wenn du deinen deye wechselrichter mit der neuesten Firmware versehen hast ( nicht das esp Gateway sondern dein WR!) und das json passt trotzdem nicht, hast du eine Deye Version die eine andere Modbus protokollversion benutzt. Dann hast jetzt 2 Möglichkeiten:
1) als Author ist @magicsven81 im Json definiert. Eventuell kann er helfen wenn du ihn anschreibst. Oder hier im Thread meldet sich jemand mit einem Deye
2) du ließt dich im Wiki ein wie man neue Modbus Items ermittelt um so die json selbst anzupassen 

Für den Fall das du eine angepasste json brauchst, bitte hier die fertige json Datei teilen ;)

Sefina-DS, 2025-03-30:
Die Firmware und das Jason scheint zu passen:

Der gesammt SoC ist mit der Position 179, 180 definiert
Laut Doku ist es dieses:

[image]

Die Register suche ich:

[image]

Gibt es eine einfache Möglichkeit die Position zu finden oder suche ich sie alleine, da ich die Werte ja kenne?

MagicSven81, 2025-03-30:
Wenn mehrere Akkus an einem Deye angeschlossen werden müssen diese als Master und Slave definiert werden.
Nur der Master sendet Daten an den Wechselrichter.
Diese Daten sind gemittelt aus allen Batteriespeichern.
Somit ist eine Abfrage eines jeden einzelnen Akkus **nicht** über den Wechselrichter möglich.

Da die Akkus aber über ein BMS verfügen (Seplos, Daly,...) können direkt am BMS die benötigten Daten (Zellspannung, Gesamtspannung, Strom, Leistung,...) über den RS485 abgegriffen werden.

Ich rufe an meinen 3 Akkupacks mit Seplos BMS je die einzelnen Werte ab.
Hierzu benötigtst du pro Akkupack einen ESP32 mit TTL-RS485 Adapter

Theoretisch wäre es auch möglich alle 3 TTL Adapter an einen ESP zu hängen.
Dazu müsste aber der Quellcode von Tobias abgeändert werden.

Sefina-DS, 2025-03-30:
Also der Deye zeigt nur die gemittelten werte an, das ist richtig, 
Die 2 Akkus sind über RS485 miteinander verbunden und der Deye redet mit beiden Akkus.
Die Werte der einzelnen Akkus kannst du auch über den Wechselrichter abfragen (Akku -> BMS) und da siehst du bis zu 15 Akkus.

In den Modbus Einstellungen sind diese ja auch angegeben. 

Wen du sagst das da ein Master und Slave gibt .... 

Wenn du sagst, ich muss 3 Adapter an je einen Akku bauen reden die Akkus nicht mehr mit dem Wechselrichter.

Wen ich es machen muss wie du es sagst, müsste ich einen TTL Adapter an den RS485 Busstrang hängen und alles abfangen.

MagicSven81, 2025-03-30:
Wie hast du deinen Akku am Wechselrichter angeschossen? CAN oder RS485?
Welches BMS verwendest du?

In meinem Fall hat das BMS 2x RS485 und 1x CAN Schnittstelle.
Du kannst am RS485 je einen Modbus anhängen. Benötigst dann aber die entsprechenden Register in der Software von Tobias.

[image]

Sefina-DS, 2025-03-30:
Was für ein BMS in den Akkus verbaut ist, kann ich nicht sage,
Die Kommunikation ist aber RS485.

Die Akkus haben die Adresse 1 und 2 und am Letzten ist noch der Abschlusswiderstand aktiv.
Die Anleitung von den Akkus:
https://www.felicitysolar.com/u_file/2503/07/file/LPBF48100-H-LPBF48200-HManual2024-12-27.pdf

Aufschrauben ist jetzt nicht so die Option.

Beim Deye schaut es so aus:

[image]

Die Akkus sind kein Selbstbau und versiegelt daher wehre es schöner die Werte aus dem Deye herauszubekommen.

MagicSven81, 2025-03-31:
@Sefina-DS  - bitte mal die genaue Bezeichnung deines Deye hier posten, dann kann ich es mir mal ansehen.

Sefina-DS, 2025-03-31:
Der Inverter ist ein : Deye SUN-12K-SG04LP3-EU 

Danke dir für deine Hilfe, habe gestern schon einiges probiert, komme aber irgendwie nicht an die Daten der Register.

MfG

MagicSven81, 2025-03-31:
> Seit dem ich das gemacht habe, obwohl die änderung valide war startet er nicht mehr .....
> 
> Ich finde es auch sehr Seltsam das der Gesammtakkustand so deviniert ist : { "position": [ 179, 180 ], "name": "BatteryCapacity", "realname": "Battery Capacity", "datatype": "integer", "unit": "%" },
> 
> [image]

Dein Screenshot gehört zu den Modbus Registern des Einphasigen Micro-Wechselrichters.
Die Register 179, 180 sind beim 12kSG04LP3 korrekt.

Sefina-DS, 2025-03-31:
Also wenn ich es richtig verstehe sind die Positionen die Byte Positionen der Antwort.
Also gehören immer 2 Positionen zu einem Register.

In deiner dokumentation (SolaxModbusGateway/tree/development/docs/ModbusProtocolReferences/deye/Deye-SUN Inverter-Modbus-en.pdf)

Steht unter Register:
[image]

Das ist (ich gehe mal davon aus) die Gesammt Akku- Werte,
unter dem Register:
[image]

Müssten doch die Werte sein, die benötigt werden ?

MfG

Sefina-DS, 2025-04-02:
Grüße,

Könntest du schon etwas raus bekommen ?

MfG

tobiasfaust, 2025-04-02:
Bin leider noch dazu gekommen, Zeit ist gerade knapp :(
Als Übergangslösung kannst du aber einen Accesspoint an das lan Kabel hängen und den ESP dann daneben. 
Nur solange bis es funktioniert

Sefina-DS, 2025-04-02:
Sorry, jetzt stehe ich gerade auf den Schlauch ?!?

MfG

tobiasfaust, 2025-04-02:
Oh, sorry. Der Kommentar gehört in einen anderen Issue. Ich kann hier in diesem Thema erstmal nichts machen.

Sefina-DS, 2025-04-25:
Da es hier wohl etwas untergegangen ist, hat, wer einen Tipp für mein Problem?

tobiasfaust, 2025-04-25:
Hier kann dir nur jemand helfen, der ebenfalls den selben Wechselrichter, und idealerweise die selbe Konstellation, wie du mit zwei Battery Packs am laufen hat. Andernfalls kannst du dir nur selbst helfen. Im Wiki ist alles ausführlich beschrieben, wie man neue Werte ermitteln und hinzufügen kann, hoffentlich auch verständlich beschrieben.

Schau mal ins Wiki unter der Rubrik: Advanced Stuff

Zb.: https://github.com/tobiasfaust/SolaxModbusGateway/wiki/adding-new-inverters-or-items

closed 2025-06-08 by tobiasfaust