tobiasfaust/SolaxModbusGateway #146: Supported by the HA solax modbus integration?

rosenrot00, 2025-03-29:
Hi,

first of all - thanks for your great work? Modbus is new to me, I was wondering if your gateway would work with this homeassistant integration. 

https://github.com/wills106/homeassistant-solax-modbus

I read it works with openWB but no idea if it would work with HA.

tobiasfaust, 2025-03-30:
Hi,
The Gateway pushes all valide as MQTT Message, so any homeautomatization Software only needs to handle those Messages.
The same for HA, only listen for mqtt messages

rosenrot00, 2025-03-30:
Thanks a lot for your response. So it would still need an integration especially when one wants to control certain settings.

I read about mqtt in HA and it seems that not much would be needed, more or less just entities per sensor or button.

closed 2025-05-15 by tobiasfaust