tobiasfaust/SolaxModbusGateway #151: Fehlerhafte Daten trotz CRC und Datasize Check

B10S-github, 2025-05-28:
Hallo :)
mir ist letztens mein ESP gestorben und das war für mich die Gelegenheit gleich mal auf die neuste Version upzudaten. Mit Freude habe ich in den Changelogs gesehen, dass das Issue [https://github.com/tobiasfaust/SolaxModbusGateway/issues/43](url) inzwischen wohl behoben wurde. Also neuste Version (3.3.3 stable) auf nen ESP, angeschlossen, MQTT/Modbus Einstellungen vorgenommen, festgestellt, dass ich relativ oft 0-Werte bekomme. Logs gecheckt und gesehen, dass ab und zu der CRC und Size Check fehlschlägt und dann nichts kommt - ok.

`[Log 4] Received CRC: 0xCE 0x16
[Log 4] Calculated CRC: 0xA9 0xFF
[Log 2] CRC check failed!
[Log 4] Dataframe length should be: 242, is: 271 bytes
[Log 2] data length check failed, should be 242 but is 271 bytes
[Log 2] Dataframe invalid`

Dann aber auch, dass CRC und Size Check ok sind, obwohl die Dataframesize eigentlich zu groß ist:

`[Log 4] Received CRC: 0xD2 0x00
[Log 4] Calculated CRC: 0xD2 0x00
[Log 4] Dataframe length should be: 130, is: 130 bytes
[Log 3] Dataframe valid, Dateframe size: 135 bytes
[Log 3] parse 135 bytes of data`

Da kommen dann 0-Werte (wobei das vermutlich Zufall ist und es auch irgendwas sein könnte) und dann auch OutOfRange Fehler:

`[Log 1] Error: position 137 out of dataframe range
[Log 1] Error: position 138 out of dataframe range`

Anbei mal der Logauszug, erst mit den Fehlerhaften CRC, dann korrekten CRC, aber trotzdem irgendwie falsch und direkt danach noch nen Fehlerfreierdurchlauf.

[SolaxMQTT_3.3.3_stable_log_20250524.txt](https://github.com/user-attachments/files/20470879/SolaxMQTT_3.3.3_stable_log_20250524.txt)

Vielleicht kann da ja wer bei Gelegenheit noch mal drüber gucken :)

MagicSven81, 2025-05-29:
Ich vermute, du verwendest den ESP an einem Deye SG04LP3?
Dann lese dir das Issues #43 nochmal genau durch. Vor allem der vorletzte Beitrag von mir.

Das springen der Werte bei einem Deye hat was mit den Anschlüssen zu tun.
Nutz den BMS Port. Da liegen die RS485 Signale sauber an.
Am Modbus Port gibt es immer wieder Unterbrechungen.

B10S-github, 2025-05-29:
Ne, läuft hinter nem SolaX X3 Hybrid.

B10S-github, 2025-05-31:
Bisher nicht. Hab jetzt aber mal ne Brücke ans PCB gelötet und werde mal gucken, ob es jetzt besser ist. Ich möchte auch nicht ausschließen, dass es bei mir ggf an der Strecke zwischen WR und Modbus Modul liegt. Baulich bedingt liegen hier 15m zwischen WR (draußen) und ESP (drinnen). Aber trotzdem hätte ich gedacht, dass dann durch den CRC und Size Check eben die fehlerhaften Pakete alle raus fliegen.
Ich werde berichten, ob es mit dem GND Anschluss jetzt besser läuft.

B10S-github, 2025-06-03:
Also mit zusätzlichem Ground läuft es auch nicht besser. Ich werde die Tage mal noch von USB-Powered auf die 13V vom WR umstellen und das ganze dann komplett über den WR versorgen, dafür muss ich aber noch irgendwie nen Buck Converter dazwischen basteln.

tobiasfaust, 2025-06-03:
Bei mir läuft es über ein 2A 5V Netzteil an einem Solax X1 Hybrid ohne Probleme. 
Hänge den ESP mal mit einem neuen und kurzen Kabel testweise dran. Die Fehler müssen aus deiner Hardwarekonfig kommen